muart-group/esphome-components #13: Support for remote temperature repeating

Sammy1Am, 2025-07-15:
Consider adding some code that allows the user to configure the frequency to send temperature updates to the heat pump, and how long to wait before timing out to the internal sensor.

Incoming remote temperatures could be recorded in a local variable, and sent to the heat pump on an interval. Still need timeout for safety.

Sammy1Am, 2025-08-02:
You all can give this a try by switching your `external_components` to the new `dev`:
```yaml
external_components:
  - source:
    type: git
    url: https://github.com/muart-group/esphome-components
    ref: dev
    components: [mitsubishi_itp]
```

And adjusting the Temperature Source configuration:
```yaml
select:
  - platform: mitsubishi_itp
    temperature_source:
      name: "Temperature Source"
      id: temp_source_select
      sources:
        - fake_temp
        - home_assistant_temperature
        - virtual_temp
      timeout: 8min
      echo_interval: 30s
```

`timeout:` Is how long before MITP explicitly switches to the internal sensor (**your heat pump may do this on its own before this**)
`echo_interval:` is how often the last reading from the current selected source will be sent to the heat pump

When a new temperature is received from an external source it will be sent immediately and the echo timer reset (so `echo_interval` is more of a maximum interval; sensors can cause messages to be sent faster than this).

milch, 2025-11-03:
Would it be possible to make the fallback sensor adjustable, at least between the internal temperature sensor and the attached thermostat? My air handler is in the basement, which in the winter is quite a bit colder than the rest of the house. Falling back to internal instead of thermostat immediately kicks the heat pump into what I like to call "hellfire" mode, which is less than desirable

Sammy1Am, 2025-11-04:
> This is what I set up

That should work fine. If the thermostat battery ever dies it'll flip-flop back and forth. I can't off-hand remember the logic but as long as selecting Thermostat *doesn't* immediately send the last value, your system would be safe. If it *does* send the last value, you'll end up permanently reporting the same temperature and reengaging hellfire mode (or hell-freezing-over mode in the summer).

> It does seem like Home Assistant only pushes sensor updates to esphome when the value changes even if the sensors report updates, which probably makes this worse.

If it's an ESPHome sensor, you can set `force_update: true` and it will always publish even if the value doesn't change, but that doesn't help for non-ESPHome sensors.

What I ended up doing is creating a service on the ESP32:
```yaml
api:
  services:
    - service: report_temperature
      variables:
        current_temperature_C: float
      then:
        - sensor.template.publish:
            id: home_assistant_temperature
            state: !lambda "return current_temperature_C;"

sensor:
  - platform: template
    id: home_assistant_temperature
    internal: true
    name: "Home Assistant"
    update_interval: never

select:
  - platform: mitsubishi_itp
    temperature_source:
      name: "Temperature Source"
      id: temp_source_select
      sources:
        - home_assistant_temperature
```

And then in Home Assistant I have an automation that pushes a temperature value to that service periodically:
```yaml
alias: Report Temperature - Basement
description: ""
mode: single
triggers:
  - entity_id:
      - sensor.env_basement_temperature
    trigger: state
conditions: []
actions:
  - data:
      current_temperature_C: "{{float(states('sensor.env_basement_temperature'))}}"
    action: esphome.env_basement_hp_report_temperature
```

This get around the Home Assistant publish frequency thing by force-pushing that value every time the automation is run. It also gives you a chance to do whatever math (like averaging several sensors) or selecting you want on the input in Home Assistant. E.g. at my house I have it report the living room temperature during the day, but switch to bedroom temperature at night.

If the sensor stops updating, you could probably catch that by modifying your automation to something like:

```yaml
actions:
  - if:
      - condition: template
        value_template: >-
          {{(now() - states.sensor.env_basement_temperature.last_updated) <
          timedelta(minutes=6)}}
    then:
      - data:
          current_temperature_C: "{{float(states('sensor.env_basement_temperature'))}}"
        action: esphome.env_basement_hp_report_temperature
    else:
      - action: select.select_option
        metadata: {}
        data:
          option: Thermostat
        target:
          entity_id: select.env_basement_hp_temperature_source
```

I haven't actually done this, but in theory it would set the source back to thermostat if the sensor stopped updating for a certain amount of time. You could even nest another `if/then` to check the Thermostat Temperature reported by the ESP32 and have the automation just do nothing if the thermostat hasn't reported anything.

milch, 2025-11-30:
The average sensor is in HASS. On the ESPHome side it is very simple:

```
# Definition: 
- platform: homeassistant
  id: downstairs_average_temp
  name: "Downstairs Average Temperature"
  entity_id: sensor.downstairs_average_temperature

# ... 

# Usage:
- platform: mitsubishi_itp
  temperature_source:
      # A list of additional temperature source sensors.  Included in this default
      # config so that Thermostat can be selected if configured.
    name: Temperature Source
    id: temp_source_select
    sources: 
      - downstairs_average_temp
    echo_interval: 60s
    timeout: 30min
```

closed 2026-03-20 by samphonic