muart-group/esphome-components #22: Heat Pump Doesn't Report Operating

Sammy1Am, 2025-08-13:
Carried over from [this issue](https://github.com/tinwer-group/mahtanar/issues/8), follow link for logs.

A PVFY-P36NAMU air handler doesn't seem to be reporting the `Operating` byte when operating. There is a branch box, but it's the only unit attached to it.

An attached Kumo adapter *does* show the unit as operating when it's operating, but it's not immediately clear if it's assuming (e.g. based on the current temperature and setpoint) or inferring it from another byte (maybe `Power` or `Actual Fan`?).

Continuing discovery/troubleshooting, I would say the next steps are:

- Go ahead and remove the Kumo from the setup (it's being noisy and sending too many packets).
- It may be harder to easily capture, but if you can capture the logs the moment the unit switches from operating to not-operating when the setpoint is reached, that might shed some light (i.e. _not_ when you make a change to mode, but when it cools down to the setpoint and turns off on its own). That would hopefully at least show which bytes *do* change between the two states.
- I also noticed that `Fan` is always reported as 0, even though Actual Fan changes.  Are you able to control the fan speed via Home Assistant?

Unfortunately a fix for this is a little unclear at the moment. If we want to set HA's ClimateAction based on a different byte, I don't know if that's scriptable in ESPHome configs, but I also don't know it wouldn't cause some other weirdness (like if we used Power, is that ever "on" when the system isn't actively heating or cooling (probably yes for defrost or preheat)).

Sammy1Am, 2025-08-19:
Hi @cameronr , just checking in to see if you think you'll be able to capture logs when the system switches (on its own) from operating to not operating).

I did a quick test (below) and confirmed that it is actually possible to set the current `action` via an ESPHome script.
```yaml
interval:
  - interval: 2min
    then:
      - lambda: |- 
          id(mitp_climate).action = climate::CLIMATE_ACTION_COOLING;
          id(mitp_climate).publish_state();
```

This means, in theory, we can write a hook on whatever attribute *does* change when it starts operating (like maybe just "power" or "actual fan") and override the `action` state so that it appears correct in Home Assistant. Being able to do this via ESPHome config is useful in this case since we may not know for sure yet which attributes to watch or how to calculate what the true current action is (but it should hopefully get you closer to being able to see the status in Home Assistant).

Actually, even if the logs are a pain to get, if you can just observe some of the different attributes either in HA or via the logs and see if any correspond to unit operating or not, I can put some sort of script together to test out.

cameronr, 2025-08-19:
Oh worry, I missed your asks in the first post. Yes, I can get some more logging, especially around when it reaches the set point on it's own.

cameronr, 2025-08-20:
Here are a bunch of logs from today:

[Archive.zip](https://github.com/user-attachments/files/21879467/Archive.zip)

I also figured out a better way to capture them (via command line instead of web interface) so it's easy to capture more / the whole day if needed.

Thanks!

cameronr, 2025-08-20:
According to HA, set temp crossed below called for temp around 20:38pm. Here's the log that includes that period.

[04.log.zip](https://github.com/user-attachments/files/21883347/04.log.zip)

Sammy1Am, 2025-08-21:
Alright, unfortunately, there doesn't appear to be any change at all (aside from the temperature dropping 0.5C) in the data reported from the beginning of that log to the end of it:

```yaml
(beginning)
Processing RunState Response: (32)[FC.62.01.30.10]09.00.00.00.02.00.00.00.00.00.00.00.00.00.00.00 52
(end)
Processing RunState Response: (96)[FC.62.01.30.10]09.00.00.00.02.00.00.00.00.00.00.00.00.00.00.00 52

Processing Settings Response: (30)[FC.62.01.30.10]02.00.00.01.03.19.00.01.00.00.00.AD.00.00.00.00 90
Processing Settings Response: (95)[FC.62.01.30.10]02.00.00.01.03.19.00.01.00.00.00.AD.00.00.00.00 90

Processing Current Temp Response: (34)[FC.62.01.30.10]03.00.00.0D.00.00.AD.00.00.00.00.00.00.00.00.00 A0
Processing Current Temp Response: (98)[FC.62.01.30.10]03.00.00.0C.00.00.AC.00.00.00.00.00.00.00.00.00 A2

Processing Error State Response: (36)[FC.62.01.30.10]04.00.00.00.80.00.00.00.00.00.00.00.00.00.00.00 D9
Processing Error State Response: (99)[FC.62.01.30.10]04.00.00.00.80.00.00.00.00.00.00.00.00.00.00.00 D9

Processing Status Response: (97)[FC.62.01.30.10]06.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 57
Processing Status Response: (97)[FC.62.01.30.10]06.00.00.00.00.00.00.00.00.00.00.00.00.00.00.00 57
```

I guess without manually checking the compressor there's no way to *confirm* for sure that it switched off during that log, but a lot of time passed 0.5C below the setpoint so it *probably* did.

I don't see any unrecognized packets being sent from the log when you had the Kumo attached, so... I hesitate to say this because it seems a little crazy, but I think the Kumo might just be *assuming* when the system is operating or not based on the current mode and current temperature / setpoint. So if the system is set to `COOL` and the temperature gets to 1C above the setpoint it will just assuming it's cooling?

The only reason this doesn't make a ton of sense is that when the Kumo boots up, if the temperature and setpoint are currently equal, it has no way to know if the system is running or not. This would be an interesting test actually:
1. Connect up the Kumo (without ESP32 is fine).
2. Set system to use the internal temperature sensor (this means the current temperature it reads will be weird, but that's not too important)
3. Set the system's setpoint so that it begins actively cooling (Kumo should say "cooling" because it knows it wasn't before)
4. When the temperature and setpoint are equal, unplug the Kumo (wait like 20 seconds) and plug it back in

In theory because the temperature is equal the Kumo won't be able to guess if it's actively cooling or not and might just default to "idle" even though the system is actively cooling. (It might also pull some shenanigans where it forces the system off so that it knows what the state is)  That's a bit of an involved test, so no pressure if this is getting to be too much work :P

**tl;dr:** There doesn't appear to be anything in the data to differentiate between operating and not-operating. I'm starting to suspect the Kumo is just guessing. The best we could do is write a script that assumes if it's warmer than the setpoint it starts to cool, and vice versa for heat; but you might be better off integrating another sensor somehow (e.g. power usage, noise, vibration, windspeed).